NSPManager/NSPanelManager #292: Color temperature resets

ehoberg, 2025-05-24:
**Describe the bug**

In my kitchen i have serval ceiling spots. 2 can simply be toggled on/off and 2 have color temperature capabilities. 
My goal was to simply set a similar color temperature and toggle them on/off.

The baviour i observed is not really correct i think. The color temperature kept resetting to ice cold. 

Changes to the configuration i've made: Enabled color temperature capabilities for that specific spot group (might be handy to mention that its a light group, not 2 individual lights. I've made this group in Zigbee2Mqtt).

After enabling this, i slide the slider to 80ish percent. After toggling the ceiling group on/off the slider reset to around 10 to 20ish percent. This kept on going. If i toggle the individual light, it works properly.

tpanajott, 2025-05-24:
Could you please set the log level of the manager to debug and the restart MQTTManager via the "Actions"-menu from the front page. Then change the lights color temperature, turn the off, then turn them on again and then send over the log? Beware that there might be sensitive information in the lot for things like MQTT and HA login. I only need the last messages regarding the changes you do with the lights, really I'm only interested in seeing what is reported back and the color temperature value.

ehoberg, 2025-05-24:
Here you go, i think this is the relevant part:

[2025-05-24 22:27:58.074] [debug] [nspanel.cpp:125] [719] Will register NSPanel 1::NSPanelKitchen relay 1 as relay
[2025-05-24 22:27:58.076] [debug] [nspanel.cpp:133] [719] Will register NSPanel 1::NSPanelKitchen relay 2 as relay
[2025-05-24 22:27:58.076] [info] [nspanel.cpp:156] [719] Loaded accepted NSPanel 1::NSPanelKitchen.
[2025-05-24 22:27:58.076] [debug] [nspanel.cpp:168] [719] Building MQTT topics for NSPanel 1::NSPanelKitchen
[2025-05-24 22:27:58.076] [info] [nspanel.cpp:205] [719] Subscribing to NSPanel MQTT topics.
[2025-05-24 22:27:58.196] [debug] [nspanel.cpp:219] [719] Loaded NSPanel 1::NSPanelKitchen, type: EU. Sending config.
[2025-05-24 22:27:58.196] [info] [nspanel.cpp:234] [719] Sending config over MQTT for panel 1::NSPanelKitchen
[2025-05-24 22:27:58.238] [debug] [nspanel.cpp:336] [719] NSPanel 1::NSPanelKitchen loading available rooms.
[2025-05-24 22:27:58.238] [debug] [nspanel.cpp:350] [719] NSPanel 1::NSPanelKitchen loaded 2 rooms.
[2025-05-24 22:27:58.238] [debug] [nspanel.cpp:358] [719] Sending updated NSPanelConfig to panel 1::NSPanelKitchen over MQTT.
[2025-05-24 22:27:58.239] [debug] [nspanel.cpp:468] [719] NSPanel 1::NSPanelKitchen became ONLINE.
[2025-05-24 22:28:12.659] [debug] [nspanel.cpp:481] [719] Got new status report from NSPanel 2::NSPanelLivingRoom
[2025-05-24 22:28:12.660] [debug] [nspanel.cpp:481] [719] Got new status report from NSPanel 1::NSPanelKitchen
[2025-05-24 22:28:13.961] [debug] [room.cpp:203] [719] Room 2:Keuken got command to turn lights on from first page.
[2025-05-24 22:28:13.961] [debug] [room.cpp:205] [719] Optimistic mode enabled, will not send state updates via callbacks in room.
[2025-05-24 22:28:13.962] [debug] [room_entities_page.cpp:124] [719] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:13.962] [debug] [room_entities_page.cpp:203] [719] Did not send update, no difference.
[2025-05-24 22:28:13.963] [debug] [room_entities_page.cpp:124] [719] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:13.963] [debug] [room_entities_page.cpp:203] [719] Did not send update, no difference.
[2025-05-24 22:28:13.964] [debug] [room_entities_page.cpp:124] [719] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:13.964] [debug] [room_entities_page.cpp:203] [719] Did not send update, no difference.
[2025-05-24 22:28:13.965] [debug] [room_entities_page.cpp:124] [719] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:13.966] [debug] [room_entities_page.cpp:203] [719] Did not send update, no difference.
[2025-05-24 22:28:13.966] [debug] [room_entities_page.cpp:124] [719] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:13.967] [debug] [room_entities_page.cpp:203] [719] Did not send update, no difference.
[2025-05-24 22:28:13.967] [debug] [room.cpp:397] [719] No table lights with color temp found, setting to 0.
[2025-05-24 22:28:13.967] [debug] [room.cpp:428] [719] Room 2::Keuken average dim level: 34, average color temperature: 81. Num ceiling lights on: 2, Ceiling lights brightness: 34, Num table lights on: 3, Table lights brightness: 34
[2025-05-24 22:28:13.967] [debug] [room.cpp:256] [719] Optimistic mode enabled, reenabled send state updates via callbacks in room.
[2025-05-24 22:28:14.217] [debug] [entity_manager.cpp:365] [724] Updating 'All rooms' status.
[2025-05-24 22:28:14.217] [debug] [entity_manager.cpp:507] [724] All rooms status updated. Waiting for next notify.
[2025-05-24 22:28:14.530] [debug] [home_assistant_light.cpp:123] [729] Got event update for HA light 22::Bar spots.
[2025-05-24 22:28:14.531] [debug] [room_entities_page.cpp:124] [729] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:14.531] [debug] [room_entities_page.cpp:203] [729] Did not send update, no difference.
[2025-05-24 22:28:14.531] [debug] [room.cpp:397] [729] No table lights with color temp found, setting to 0.
[2025-05-24 22:28:14.531] [debug] [room.cpp:428] [729] Room 2::Keuken average dim level: 34, average color temperature: 81. Num ceiling lights on: 2, Ceiling lights brightness: 34, Num table lights on: 3, Table lights brightness: 34
[2025-05-24 22:28:14.781] [debug] [entity_manager.cpp:365] [724] Updating 'All rooms' status.
[2025-05-24 22:28:14.781] [debug] [entity_manager.cpp:507] [724] All rooms status updated. Waiting for next notify.
[2025-05-24 22:28:17.071] [debug] [room.cpp:203] [719] Room 2:Keuken got command to turn lights on from first page.
[2025-05-24 22:28:17.071] [debug] [room.cpp:205] [719] Optimistic mode enabled, will not send state updates via callbacks in room.
[2025-05-24 22:28:17.072] [debug] [room_entities_page.cpp:124] [719] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:17.072] [debug] [room_entities_page.cpp:203] [719] Did not send update, no difference.
[2025-05-24 22:28:17.073] [debug] [room_entities_page.cpp:124] [719] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[... cut ...]
[2025-05-24 22:28:23.297] [debug] [room_entities_page.cpp:196] [719] Publishing state update to nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:28:23.299] [debug] [room.cpp:397] [719] No table lights with color temp found, setting to 0.
[2025-05-24 22:28:23.299] [debug] [room.cpp:428] [719] Room 2::Keuken average dim level: 34, average color temperature: 0. Num ceiling lig

What might be the cause is the fact that there is no avarage color temperature for table lights:
[2025-05-24 22:28:17.641] [debug] [room.cpp:397] [729] No table lights with color temp found, setting to 0.

tpanajott, 2025-05-24:
That is indeed the correct part of the log but unfortunately it seems I've moved that part of logging to trace level.

The table lights having 0 color temperature shouldn't really matter as the color temperature is only calculated from lights that are on and can have color temperature.

ehoberg, 2025-05-24:
I redid the test and downloaded the trace logging.

The relevant entity is:
entities/lights/22/state' -> 'Bar spots"(2028'

2025-05-24 22:55:27.512] [trace] [command_manager.cpp:16] [828] Received command from MQTT: 
d 8
[2025-05-24 22:55:27.512] [trace] [command_manager.cpp:20] [828] Calling callbacks.
[2025-05-24 22:55:27.512] [debug] [room.cpp:203] [828] Room 2:Keuken got command to turn lights on from first page.
[2025-05-24 22:55:27.512] [debug] [room.cpp:205] [828] Optimistic mode enabled, will not send state updates via callbacks in room.
[2025-05-24 22:55:27.512] [trace] [light.cpp:210] [828] Requested light 7::Keuken spots to be color temperature 2000.
[2025-05-24 22:55:27.512] [trace] [home_assistant_manager.cpp:191] [828] [HA WS] Sending data: {"domain":"light","id":12,"service":"turn_on","service_data":{"kelvin":2000},"target":{"entity_id":"light.kitchen_spots"},"type":"call_service"}
[2025-05-24 22:55:27.512] [trace] [light.cpp:121] [828] Sending state update to MQTT for light 7::Keuken spots
[2025-05-24 22:55:27.512] [trace] [mqtt_manager.cpp:286] [828] Publising 'nspanel/mqttmanager_192.168.1.94/entities/lights/7/state' -> '
Keuken spots" d(2028'
[2025-05-24 22:55:27.512] [debug] [room_entities_page.cpp:124] [828] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:55:27.513] [debug] [room_entities_page.cpp:203] [828] Did not send update, no difference.
[2025-05-24 22:55:27.515] [trace] [mqtt_manager_config.cpp:40] [827] Found setting all_rooms_status_backoff_time with value 250
[2025-05-24 22:55:27.515] [trace] [light.cpp:210] [828] Requested light 22::Bar spots to be color temperature 2000.
[2025-05-24 22:55:27.515] [trace] [home_assistant_manager.cpp:191] [828] [HA WS] Sending data: {"domain":"light","id":13,"service":"turn_on","service_data":{"kelvin":2000},"target":{"entity_id":"light.old_kitchen_spots"},"type":"call_service"}
[2025-05-24 22:55:27.515] [trace] [light.cpp:121] [828] Sending state update to MQTT for light 22::Bar spots
[2025-05-24 22:55:27.515] [trace] [mqtt_manager.cpp:286] [828] Publising 'nspanel/mqttmanager_192.168.1.94/entities/lights/22/state' -> '
	Bar spots" d(2028'
[2025-05-24 22:55:27.515] [debug] [room_entities_page.cpp:124] [828] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:55:27.516] [debug] [room_entities_page.cpp:203] [828] Did not send update, no difference.
[2025-05-24 22:55:27.516] [trace] [light.cpp:210] [828] Requested light 23::Spiegel lamp to be color temperature 2000.
[2025-05-24 22:55:27.516] [trace] [home_assistant_manager.cpp:191] [828] [HA WS] Sending data: {"domain":"light","id":14,"service":"turn_on","service_data":{"kelvin":2000},"target":{"entity_id":"light.wall_mirror_lamp"},"type":"call_service"}
[2025-05-24 22:55:27.516] [trace] [light.cpp:121] [828] Sending state update to MQTT for light 23::Spiegel lamp
[2025-05-24 22:55:27.516] [trace] [mqtt_manager.cpp:286] [828] Publising 'nspanel/mqttmanager_192.168.1.94/entities/lights/23/state' -> '
Spiegel lamp" d(202'
[2025-05-24 22:55:27.516] [debug] [room_entities_page.cpp:124] [828] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:55:27.516] [debug] [room_entities_page.cpp:203] [828] Did not send update, no difference.
[2025-05-24 22:55:27.516] [trace] [light.cpp:210] [828] Requested light 20::Stolp serre to be color temperature 2000.
[2025-05-24 22:55:27.517] [trace] [home_assistant_manager.cpp:191] [828] [HA WS] Sending data: {"domain":"light","id":15,"service":"turn_on","service_data":{"kelvin":2000},"target":{"entity_id":"light.stolp_serre"},"type":"call_service"}
[2025-05-24 22:55:27.517] [trace] [light.cpp:121] [828] Sending state update to MQTT for light 20::Stolp serre
[2025-05-24 22:55:27.517] [trace] [mqtt_manager.cpp:286] [828] Publising 'nspanel/mqttmanager_192.168.1.94/entities/lights/20/state' -> '
Stolp serre" d(202'
[2025-05-24 22:55:27.517] [debug] [room_entities_page.cpp:124] [828] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:55:27.517] [debug] [room_entities_page.cpp:203] [828] Did not send update, no difference.
[2025-05-24 22:55:27.517] [trace] [light.cpp:210] [828] Requested light 21::Stolp bar to be color temperature 2000.
[2025-05-24 22:55:27.517] [trace] [home_assistant_manager.cpp:191] [828] [HA WS] Sending data: {"domain":"light","id":16,"service":"turn_on","service_data":{"kelvin":2000},"target":{"entity_id":"light.stolp_bar"},"type":"call_service"}
[2025-05-24 22:55:27.517] [trace] [light.cpp:121] [828] Sending state update to MQTT for light 21::Stolp bar
[2025-05-24 22:55:27.517] [trace] [mqtt_manager.cpp:286] [828] Publising 'nspanel/mqttmanager_192.168.1.94/entities/lights/21/state' -> '
	Stolp bar" d(202'
[2025-05-24 22:55:27.518] [debug] [room_entities_page.cpp:124] [828] Entity page 1 sending state update to MQTT topic: nspanel/mqttmanager_192.168.1.94/entity_pages/1/state
[2025-05-24 22:55:27.518] [debug] [room_entities_page.cpp:203] [828] Did not send update, no difference.
[2025-05-24 22:55:27.518] [debug] [room.cpp:397] [828] No table lights with color temp found, setting to 0.
[... cut ...]
[2025-05-24 22:56:03.768] [trace] [mqtt_manager.cpp:257] [828] Processing message from topic nspanel/E4:65:B8:9B:C9:F8/log
[2025-05-24 22:56:03.769] [trace] [mqtt_manager.cpp:257] [828] Processing message from topic nspanel/E4:65:B8:9B:C9:F8/log
[2025-05-24 22:56:03.769] [trace] [mqtt_manager.cpp:257] [828] Processing message from topic nspanel/E4:65:B8:9B:C9:F8/log
[2025-05-24 22:56:03.769] [trace] [mqtt_manager.cpp:257] [828] Processing message from topic nspanel/E4:65:B8:9B:C9:F8/log
[2025-05-24 22:56:03.769] [trace] [mqtt_manager.cpp:257] [828] Processing message from topic nspanel/E4:65:B8:9B:C9:F8/log

tpanajott, 2025-05-24:
Thanks! What does HA say the Kelvin is for that light? I think you can see the actual number by checking the entity under developer tools in HA.

ehoberg, 2025-05-24:
I redid the test to observe HA. I moved the slider all the way up. Checked HA, toggled on, and off.

All the way up:

color_temp_kelvin: 2202
color_temp: 454

After toggle:

color_temp_kelvin: 5000
color_temp: 200

The interesting part is, the fact that i had to read that out of the individual lights, instead of the group. The group itself does not have the actual current value from the looks of it

[image]

tpanajott, 2025-05-24:
> I redid the test to observe HA. I moved the slider all the way up. Checked HA, toggled on, and off.
>
> All the way up:
>
> color_temp_kelvin: 2202
> color_temp: 454
>
>
> After toggle:
>
> color_temp_kelvin: 5000 [...]
That's interesting. Given that the manager will send the command to set the color temp and (assuming you have changed it from optimistc mode) it will assume the correct brightness and color temp was set. But, as there's no color temp value when it gets the feedback it can't really know what color temp or really is. Interestingly my zigbee2mqtt group of lights (4 IKEA color temp E27) does contain color temp.[image]

ehoberg, 2025-05-24:
I made a small mistake the group does contain it, i was looking at the wrong node. 
That being said, i maybe did find out something else thats interesting. 
If toggled off, the current temp is null. So if a readback is done while its off, it might explain the reset?

[image]

The spots im using are INNR lights:
https://www.zigbee2mqtt.io/devices/RS_227_T.html#innr-rs%2520227%2520t

tpanajott, 2025-05-24:
Yeah, might be though om pretty sure it would cause an exception and not be registered. But, does the group contain the expected value or the value that you see on the panel?

ehoberg, 2025-05-25:
I tried again this morning.

I've set the sliders all the way up. Observed value in HA, it was set correctly. Then I toggled it off and on via the ceiling group and it got reset to a lower value. 

What I observed from the light itself is that it was set to on with the correct color value before it got reset to a cold temperature. 

When I did the same with the individual light control it behaved occordingly. Values were correct when I toggled it on and off. When I switched back to the ceiling control, toggling it on and off, it actually worked O_o. Now I can't reproduce it anymore, for some reason it fixed itself.

That being said, when I toggle the ceiling on/off sometimes not all lights behave accordingly. The group says it's off, the individual light as well, but it's actually on.

Something strange is going on

ehoberg, 2025-07-18:
I have not been able to reproduce it. At some point it just stopped happening.

I do get an occasional non response when I use the master slider for brightness. The spots sometimes ignore the changes, while the filament lamp just works. It might be related, however I think we can close this ticket.

tpanajott, 2025-07-18:
Sounds good to me. The issue you are describing might be related to #297.

closed 2025-07-18 by tpanajott