NSPManager/NSPanelManager #293: A few cosmetic issues.

aaronouthier, 2025-05-26:
**Describe the bug**
Two display anomalies: one on the configuration screen, and one on the temperature readout.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Entities'
2. Click on 'Weather & Time'
3. Scroll down to 'Wind Speed Format'
4. See 4 choices: 
   Km/h
   Meters/s
   Mph
   Mph

**Expected behavior**
Formatting should be consistent (sorry, a bit nit-picky here): Either Kmph & Mps, **OR** M/h. Variation in local convention may override this, as the difference may be expected. Also, maybe change second Mph to Ft/s or eliminate the redundancy altogether.

**Environment info (please complete the following information):**
 - Home Assistant user
 - Browser Microsoft Edge
 - NSPanelManager version [No Version Listed].
 - NSPanel Firmware version 0.3.593

**Additional context**
Separate issue: My internal temperature sensor is showing constant 20.4 degrees Fahrenheit. That's below freezing!! Been that way for over a month, despite upgrading NSPM version and updating Firmware this morning.

aaronouthier, 2025-05-26:
I just pulled the latest git source this morning before making this report and re-ran your docker script. For some reason, the version number doesn't show anywhere. When I look under "About", I get:
----
Version: %version%
NSPanel Manager is a software for easy management of multiple Sonoff NSPanels and intuitive use for all family members for the whole house.
----

Running on a x86-64 KVM instance inside of Proxmox.

Note that I did have to recreate the VM after my boot NVMe SSD went out on me. The NSPM install is fresh as of about a month ago, but the panel install is almost a year running. The timing of the temperature behavior change does coincide with the rebuilding of the manager from scratch. Perhaps an old config setting is "stuck"?

Also, not sure if directly related: I'm getting the following in the logs:
`
08:57:48 AM | WARNING | (2189208) httpd_txrx: httpd_resp_send_err: 404 Not Found - Nothing matches the given URI
-- | -- | --
08:57:48 AM | WARNING | (2189208) httpd_uri: httpd_uri: URI '/favicon.ico' not found
08:21:31 AM | WARNING | (11778) ButtonManager: No config currently loaded/set. Unable to determine if received message was from a relay group. Will not change any relay state.
08:21:31 AM | WARNING | (11738) ButtonManager: No config currently loaded/set. Unable to determine if received message was from a relay group. Will not change any relay state.
08:21:30 AM | ERROR | (11488) ButtonManager: Failed to get current config while trying to subscribe/unsubscribe from relay group topics.
08:21:30 AM | WARNING | (11468) event: handler already registered, overwriting
08:08:49 AM | WARNING | (11868) ButtonManager: No config currently loaded/set. Unable to determine if received message was from a relay group. Will not change any relay state.
08:08:48 AM | WARNING | (11838) ButtonManager: No config currently loaded/set. Unable to determine if received message was from a relay group. Will not change any relay state.
08:08:48 AM | ERROR | (11418) ButtonManager: Failed to get current config while trying to subscribe/unsubscribe from relay group topics.
08:08:48 AM | WARNING | (11418) event: handler already registered, overwriting
`

aaronouthier, 2025-05-29:
I switched to official Docker interface, and reflashed the firmware.
In the future, a way to factory-reset a given panel might help with situations where the manager goes down and has to be replaced. I couldn't ever get an existing panel to resync with a new manager. I'm guessing this is something you haven't tested before?

tpanajott, 2025-05-29:
> I switched to official Docker interface, and reflashed the firmware.
> In the future, a way to factory-reset a given panel might help with situations where the manager goes down and has to be [...]

They should simply show up available to accept if connected to the same MQTT server and restarted.

aaronouthier, 2025-05-29:
It did indeed show up, and I accepted them. However, error messages about not being able to overwrite existing config or something similar would show in the panel log. Also, the temperature readout stopped working, and I was unable to reboot or flash the panel from the manager - the panel wouldn't respond to such requests.

I should note that although I own 2 panels, only one is connected and working. That is my test dummy for now. It works great now, after the reflash. Because I have only 1 panel, I realize this could've just been a fluke. Still more testing to do