NSPManager/NSPanelManager #300: Factory reset button.

aaronouthier, 2025-05-29:
I'd love to have a factory reset option, to clear all config data without needing to reflash the panel via cable.

Also, perhaps a reboot/restart button. Both could be available inside WebUI of the individual panel. Might never be necessary, but better to have it and not need it, than to need it and not have it.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

aaronouthier, 2025-05-29:
Interesting, Issue #300 was assigned. I believe I created #200 also.

Not relevant to the issue, but interesting nonetheless.

tpanajott, 2025-07-13:
This should be fixed as of latest beta when we release that.

tpanajott, 2025-07-15:
Fixed in latest beta release v2.0.9. Closing.

closed 2025-07-15 by tpanajott