NSPManager/NSPanelManager #309: template button

ungil-73, 2025-06-22:
Hello,
I have an Avatto roller shutter controller integrated into Home Assistant via Zigbee2MQTT.
I've created template buttons in Home Assistant to control cover.open, cover.close, and cover.stop functions. These buttons work fine on the Home Assistant interface.
Is it possible for NSPanel Manager to handle these buttons as well — or even control the cover entities directly — instead of only supporting switches? Or is it already possible, and I might just be doing something wrong?

Thanks in advance!

cablesandcoffee, 2025-07-13:
Hi sorry late answer. only switches are implemented as of yet. Blinds page to control blinds is on the roadmap. You could solve it for now by creating specific switches which triggers home assistant scenes where the blinds move to the desired position. ON could trigger an UP scene and OFF could trigger Down scene for example. When they're moving you could have a switch that turns ON and if you turn that OFF with the panel the Blinds would stop. Not the best solution bot for now that is whats possible.

tpanajott, 2025-07-16:
Just implemented this and NSPM will be able to handle "button.*" and "input_button.*" in the next release as well as manual MQTT topic + payload. We simply need an icon from @cablesandcoffee to display on the panel so the user known where to press.

cablesandcoffee, 2025-07-18:
Updated all tft files, both EU and US with new fonts containing button press icon. It's under lowercase z.

[image]

tpanajott, 2025-07-18:
Looks good, icon implemented and ready for release.[image]

closed 2025-07-31 by tpanajott