NSPManager/NSPanelManager #312: After New update, nspmanager stopped connecting to mqtt

mosphorus, 2025-07-15:
On website I can't see any panels. Only message that it is connecting to mqttmanager

Go to nspanelmanager web

I should see my panel status and there is nothing from it

[2025-07-15 12:11:35.762] [debug] [websocket_server.cpp:133] [118] Websocket connected. URL: /websocket/stomp
[2025-07-15 12:11:35.769] [debug] [websocket_server.cpp:144] [118] Websocket closed. Code: 1006, Reason: Abnormal closure

HA on pc. I'm using opera mainly.

mosphorus, 2025-07-15:
Panel itself is working with delays - probably according the log, it is connected but nspanelmanager Web side, is disconnecting

mosphorus, 2025-07-15:
I see it now as well. Its only on opera. Used Firefox and all is good. Performed even update on panel without an issue.

tpanajott, 2025-07-15:
New release out now that should fix this. Please retry and come back if it's fixed or not.

tpanajott, 2025-07-15:
Nice! It was probably an issue on all chromium-based browsers.

mosphorus, 2025-07-15:
Thanks, I can confirm that it works on my browser correctly now.