NSPManager/NSPanelManager #315: Screen Save Brightness Not Persisted

beerygaz, 2025-07-20:
At 8pm I set a panel screen saver brightness to 0%.  This turns off the screensaver.  At 7am i set it back to 50% so it works in daylight.  However, if someone uses the panel after the 0 value setting has been activated, the panel reverts to the default (50%) brightness. 

I use the `number.<panelname>_screensaver_brightness` entity to control this value in an automation. 

Not sure if this is a bug or expected behaviour.

beerygaz, 2025-07-20:
Spoke too soon.  Saw the screen invert but did not do proper testing.  The ‘touch’ area of the display don’t seem to correlate to the images.  The ‘scenes’ icon does not work and the room dropdown doesn’t open the entities page.  Will do some more testing this week to try and provide some more info.

tpanajott, 2025-07-20:
> Spoke too soon. Saw the screen invert but did not do proper testing. The ‘touch’ area of the display don’t seem to correlate to the images. The ‘scenes’ icon does not work and the room dropdown doesn’t open the entities page. Will do some more testing this week to try and provide some more info.

Perhaps this was meant for your other issue?

beerygaz, 2025-07-21:
It was indeed.  Apologies.   

Here is some more testing on this issue

MQTT events for the device once the automation has run, clearly setting screen brightness and screensaver brightness to 0.

[image]

Settings from the the panel settings page in NSPanelManager::

[image]

Panel is currently behaving as expected.   Screen is "off" (barring a few fingerprints)

[image]

After tapping the screen to wake it up, however, after the screensaver timeout period it reverts to default brightness (60 per the settings page)

[image]

Version Information
Version on panel web UI: Version 0.1.152
HA Add-On version: 1.0.4
HA 2025.7.2
HAOS - 15.2