NSPManager/NSPanelManager #318: Rework swipe functionality

cablesandcoffee, 2025-07-27:
Firmware and tft file need some rework to perfect the touch event detection function in the tft file.
The main thing is that we want the tft file to detect what kind of touch event that has happened and send that information to the firmware.
Currently a long press for example is calculated in the firmware and we want to change that so all that is done by the nextion screen in the tft file.

In this topic we could make a list of all the things that will be affected when doing this rework.

tpanajott, 2025-07-27:
I think this would be an excellent opportunity to also rework the slider/progressbar shown when saving a scene on the scenes page. The "hold to save" logic could be moved into the Nextion MCU to allow for a smoother  progressbar when saving and then once a save is complete, simply sending a code we agree on. Something like 0x90 (for saved scene) and then the button ID, so like a total command would be 0x9001 for button one.

It would also be nice if we could move the whole "enter only ceiling/table lights mode" logic into the Nextion panel and then simply send something like 0x91 for normal mode, 0x92 for "table lights mode" and 0x93 for "ceiling lights mode" or something like that.

cablesandcoffee, 2025-08-04:
Yeah totally agree. I would like to suggest that we switch to using print instead of printh in nextion aswell. So instead of sending weird hexadecimal stuff we send our own commands in plain text like "shortpress", "longpress" "savedone" or something like that.