NSPManager/NSPanelManager #323: Can't access 192.168.4.1 !

LukasBehr, 2025-08-26:
**Describe the bug**
I have followed the steps on the website carefully, and I am stuck here [https://nspanelmanager.com/docs/getting-started/configure-flashed-nspanel]. Unfortunately, I can not access the 192.168.4.1 website. I have already tried it on my laptop and two smartphones.. I have also reflashed it again, but nothing has changed. Any help is highly appreciated!

**To Reproduce**
After flashing your NSPanel:

    1 .Connect it to power
    2. It will boot as an access point.
    3. Check for a WIFI network called NSPMPanel-XXXXXX where XXXXXX is unique to each panel.
    4. Connect to it (no password required)
    5. Visit http://192.168.4.1/ -> ERROR

**Expected behavior**
I hoped for seeing the website :(

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment info (please complete the following information):**
 - Browser [e.g. chrome]

**Additional context**
Add any other context about the problem here.

tpanajott, 2025-08-26:
So, that indicates that you have installed the stable version of the firmware. The 4.1 address is fot the beta firmware. Either reinstall with the beta firmware (if you are running beta manager) or simply use 192.168.1.1 to access the setup page.

tpanajott, 2025-08-26:
You need them both to be configured for the same MQTT server/broker. What version of the manager are you running?

LukasBehr, 2025-08-26:
> You need them both to be configured for the same MQTT server/broker. What version of the manager are you running?

1.0.4

tpanajott, 2025-08-26:
Hmm, so both are running stable. Then it should just automatically popup. Try restarting the manager. Also, verify that you nspanel actually connected to your wifi.

LukasBehr, 2025-08-26:
> Hmm, so both are running stable. Then it should just automatically popup. Try restarting the manager. Also, verify that you nspanel actually connected to your wifi.

It appears not to be connected to the wifi because there is an exclamation mark next to the wifi symbol displayed on the top right corner of the ns panel screen. However the NSPMPanel wifi is gone...

tpanajott, 2025-08-26:
Yeah, that to be expected. That's not the GUI file we use so you can simply ignore everything on the screen until you have our NSPanelManager file flashed. You can check in your router if the panel has connected to the wifi. If it hasn't then it should open the WiFi access point again after 5-10 minutes (can't remember exactly, think it's 5) so you can configure it again.

LukasBehr, 2025-08-26:
> Yeah, that to be expected. That's not the GUI file we use so you can simply ignore everything on the screen until you have our NSPanelManager file flashed. You can check in your router if the panel [...]

Apparently it is connected! However a restart of the manager did not help...

[image]

tpanajott, 2025-08-26:
Hm, strange. Perhaps check you logs for the MQTT server to see if it connects correctly?

If you need to reconfigure the NSPanel you can simply enter the IP address of it in a browser that's connected to your WiFi.

LukasBehr, 2025-08-26:
> Hm, strange. Perhaps check you logs for the MQTT server to see if it connects correctly?
> 
> If you need to reconfigure the NSPanel you can simply enter the IP address of it in a browser that's connected to your WiFi.

That seems to be the problem: 

[image]

LukasBehr, 2025-08-26:
> Ah, then either you manager or your panel has failed to connect to MQTT. How are you running the MQTT server/broker? HA addon?

Yes, I am running with HA addon and put in the following information: 

[image]

tpanajott, 2025-08-26:
Ah, remove "mqtt://" It should only be the server name/IP.

LukasBehr, 2025-08-26:
Ah thank you! But we are still stuck here:

[image]

I assume I don't wrote down the port instead of the MQTT iP. Wehere can I find the MQTT IP?

aaronouthier, 2025-08-26:
Same as your home assistant IP address if running as an HAOS add-on.

tpanajott, 2025-08-26:
> Same as your home assistant IP address if running as an HAOS add-on.

Thanks, if that doesn't work you should be able to run `nslookup mymqttserveraddress` on Windows or `dig mymqttserveraddress` on Mac. Though, don't quote me on it as I don't run either. Obviously remove "mymqttserveraddress" with whatever you address is.

You should enter simply the IP without port and such, so no trailing :1883 at the end.

LukasBehr, 2025-08-26:
> Thanks, if that doesn't work you should be able to run `nslookup mymqttserveraddress` on Windows or `dig mymqttserveraddress` on Mac. Though, don't quote me on it as I don't run either. Obviously [...]

I did, but it still does not work...

aaronouthier, 2025-08-26:
Ok, so home assistant has a built-in mqtt broker, but that isn't accessible to other things outside of HAOS. If you aren't already, make sure you are running an MQTT **ADD-ON** and not trying to use the built-in HAOS mqtt broker.

The only other thing I can think of, is to make sure you setup authentication. I use the Mosquitto add-on, and I have a separate user configured that I use for MQTT stuff, but you can also use your HA login info.

LukasBehr, 2025-08-27:
> Hmm, not sure, but try re-entering the authentication info and saving. Perhaps it somehow got cleared by all of the previous edits?

I have reentered it, but no success... Should I set up credentials in my MQTT add-on and try it?

[image]

tpanajott, 2025-08-27:
Unfortunately I really do now know that much about the Mosquito addon for HA as I don't use it myself. Are you sure its the manager that is stuck and giving the warning of a not authenticated user?

Did you change login information in the manager and the panel?

aaronouthier, 2025-08-27:
I'm still trying to get my Landlord (my Dad) to put my 2 NSPanels up. I had one up before, but I moved into some new construction here on the property, and had to take it down. I do miss my panels. I only had one up previously, but I own two.

LukasBehr, 2025-08-27:
> [@tpanajott](https://github.com/tpanajott) Nice catch, Tim. There are 2 places that need to have the MQTT info entered, not just one! [@LukasBehr](https://github.com/LukasBehr) Check whether the IP address mentioned in the MQTT logs is that of the manager or the panel. Whichever one, double-check that it has the correct credentials.

Thanks for the input! 

1. Should the NSPanelManager address be the same like the IP address of the HA?
2. 2025-08-27 19:56:37: New connection from 172.*****1:57906 on port 1883. The connection comes from a strange IP starting with 172... That is a little weird...

LukasBehr, 2025-08-27:
IT FINALLY WORKED! I have messed up the credentials in the NSPManager .....

THANK YOU VERY MUCH GUYS!!! YOU WERE A HUGE HELP! YOU ARE AWESOME! :)

aaronouthier, 2025-08-27:
Correct, if you have the NSPManager installed as an Add-on in Home Assistant. It's probably the easiest way to install it. I have mine running on a separate machine.