NSPManager/NSPanelManager #324: "Manager has no checksum for installed firmware on panel"

erich74, 2025-08-31:
Hello, I just flashed my NSPanel with the web flasher (beta v2).
It is connecting to Home Assistant running in a VM on ProxMox and Mosquito mqtt is running in another VM.
The NSPanel is properly linked to HA and MQTT but in the NSPanel Manager I have the following warning 

Manager has no checksum for installed firmware on panel.
Manager has no checksum for installed GUI on panel.

Environment info
 - Home Assistant
 - Browser chrome
 - NSPanelManager version 0.3.613
 - NSPanel Firmware Version: 2.1.1-beta-h1

tpanajott, 2025-08-31:
I'm guessing this is just a recently flashed panel that hasn't had the GUI updated to the NSPanel Manager GUI I guess? If so, that explains the lack of checksum for installed GUI on the panel.

Regarding the missing checksum for installed firmware you can simply do a firmware update from the manager and it should resolve itself.

Basically, when you flash the panel it doesn't know it's own checksum. When you flash the panel from the manager it will flash the new firmware, reboot, then check if the firmware update was successful and if so, download the current checksum from the manager and store it internally and report it back to the manager periodically.

erich74, 2025-08-31:
Hi, thanks a lot for your answer however I am not sure I fully understand your explanation.
Yes this is a newly flashed deviced, I used the web flasher (beta v2 version). 
The screen is blank and I have those messages in the manager:

Manager has no checksum for installed firmware on panel.
Manager has no checksum for installed GUI on panel.

In the manager if I go to my device settings and I select either "Update firmware" or "update GUI" nothing happens and I get the 2 warning messages above.

I don't really understand what I should do next.

I understand the firmware itself is not enough and I need to flash the GUI but I don't know how to do it.

I have followed the online instructions here: https://nspanelmanager.com/ but it is not talking about the GUI flashing 

I just used this link https://nspanelmanager.com/flashing-tool.html

But I guess this is the NSPanel firmware but not the GUI flash, correct?

Thanks for your help

erich74, 2025-08-31:
Thanks again for your time.

Here are some screenshots. Everything looks fine but clicking flash gui does not do anything

[image]
[image]
[image]

tpanajott, 2025-08-31:
Ah, well, that 172.30.32.1 does seem like an IP for a docker NAT network. What is the IP of the proxmox VM you are running it on? That's the IP you're supposed to enter. Docker handles all of the network stuff to translate from your VM IP to the container itself.

erich74, 2025-08-31:
HA is running in its own VM (no docker) its IP is 192.168.1.126
MQTT is running in docker on another VM. The IP of its VM is 192.168.1.126 and the IP of the docker container running MQTT is 172.18.0.2

I don't really understand where this 172.30.32.1 is coming from. Initially I entered 192.168.1.126 which is the IP of the VM where Home Assistant is running with the NSPanel Manager BETA

tpanajott, 2025-08-31:
Ahh, are you running it as an HA addon?

erich74, 2025-08-31:
yes

tpanajott, 2025-08-31:
Ah, then the manager address is the IP you use to access Home Assistant. Ha itself does run all addons as docker containers.

erich74, 2025-08-31:
ok I'm so sorry I think I have clicked on the "load from url" link and saved and this is when I got this 172.30.32.1 IP.
No I have reverted to the HA IP and GUI is updating.
It's not finished yet but I guess problem is solved.
I will confirm

Thanks a million for your patience!

tpanajott, 2025-08-31:
Hey, no worries. Glad you got it running.

tpanajott, 2025-08-31:
Also, it is really prone to interference so thats all fun. It might take a couple of tries but it should show a progress bar once its starting to update.

erich74, 2025-08-31:
do you know how I could completely wipe the content of the board including the GUI part?

erich74, 2025-08-31:
I am trying flashing the GUI @ 9600 bauds but no luck

[image]

erich74, 2025-09-01:
Add-on log you mean the log from Home Assistant add-on ? If so no I didn't . I'll reflash and check it.

Sorry for the stupid question but how do you chose the GUI file to upload in NSPanel manager ? 
Panel type & orientation there's only EU for the EU version. 

Thanks again for your support

tpanajott, 2025-09-01:
Press the panel name on the first page (to get to where ypy saw the panel log) and press the preview. 

https://nspanelmanager.com/docs/web-interface#choose-between-different-backgrounds-for-the-panel

erich74, 2025-09-01:
oh my g... I never thought you could click on the image to get a preview :)

I changed the display and reflashed GUI but still the same.
Here is the HA add-on log from when I started the NSPanel (powered on) until it fininshed flashing GUI

[2025-09-01 20:27:02.156] [info] [main.cpp:37] [33] Reloading config from manager. Will wait 50ms for other processing to complete before proceeding with config reload.
[pid: 37|app: 0|req: 23/78] 192.168.1.87 () {52 vars in 1164 bytes} [Mon Sep  1 18:27:02 2025] POST /save_nspanel_settings/2 => generated 0 bytes in 39 msecs (HTTP/1.1 302) 7 headers in 236 bytes (1 switches on core 0)
[pid: 36|app: 0|req: 14/79] 192.168.1.87 () {46 vars in 973 bytes} [Mon Sep  1 18:27:02 2025] GET /nspanel/2 => generated 162777 bytes in 44 msecs (HTTP/1.1 200) 8 headers in 366 bytes (1 switches on core 0)
[2025-09-01 20:27:02.207] [info] [mqtt_manager_config.cpp:49] [33] Read timezone Europe/Paris from /etc/timezone.
[2025-09-01 20:27:02.207] [info] [mqtt_manager_config.cpp:52] [33] Loading MQTT Manager settings.
[2025-09-01 20:27:02.211] [info] [mqtt_manager_config.cpp:124] [33] Updating/calculating MD5 checksums for all firmware files.
[2025-09-01 20:27:02.213] [debug] [websocket_server.cpp:147] [269] Websocket closed. Code: 1011, Reason: Internal error
[2025-09-01 20:27:02.214] [info] [mqtt_manager_config.cpp:130] [33] Firmware checksum: caac1ce43da0b8ec7e8883f038ba4ff7
[2025-09-01 20:27:02.215] [info] [mqtt_manager_config.cpp:140] [33] LittleFS checksum: c87ef0724cee537c988dbd3ccf5ca45b
[2025-09-01 20:27:02.215] [info] [mqtt_manager_config.cpp:149] [33] Updating/calculating MD5 checksums for all .tft GUI files.
[2025-09-01 20:27:02.241] [info] [mqtt_manager_config.cpp:155] [33] EU TFT1 GUI file checksum: 8209ac6badc6560cb86c38ea795350aa
[2025-09-01 20:27:02.267] [info] [mqtt_manager_config.cpp:165] [33] EU TFT2 GUI file checksum: cb51b3e3e2deb8116af02ff9fd94ddb8
[2025-09-01 20:27:02.293] [info] [mqtt_manager_config.cpp:175] [33] EU TFT3 GUI file checksum: c1c8404e4848365132cadd56e56c60c5
[2025-09-01 20:27:02.318] [info] [mqtt_manager_config.cpp:185] [33] EU TFT4 GUI file checksum: 74f1abbb1cfd56d78d79e4f4e230e14d
[2025-09-01 20:27:02.338] [info] [mqtt_manager_config.cpp:195] [33] US TFT1 GUI file checksum: 8889645f2bd0cdceb67226687be58bd7
[2025-09-01 20:27:02.347] [debug] [websocket_server.cpp:136] [270] Websocket connected. URL: /websocket/stomp.
[2025-09-01 20:27:02.358] [info] [mqtt_manager_config.cpp:205] [33] US TFT2 GUI file checksum: 39b55d1f0037a31e747d1de6355dad47
[2025-09-01 20:27:02.365] [debug] [websocket_server.cpp:169] [270] Got STOMP connection request, accepting request, sending CONNECTED
[2025-09-01 20:27:02.368] [debug] [websocket_server.cpp:218] [270] STOMP subscribing to topic nspanel/C0:49:EF:D3:A1:D8/status (id: sub-0)
[2025-09-01 20:27:02.368] [debug] [websocket_server.cpp:218] [270] STOMP subscribing to topic mqttmanager/events (id: sub-1)
[2025-09-01 20:27:02.368] [debug] [websocket_server.cpp:218] [270] STOMP subscribing to topic mqttmanager/warnings (id: sub-2)
[2025-09-01 20:27:02.369] [debug] [websocket_server.cpp:218] [270] STOMP subscribing to topic nspanel/C0:49:EF:D3:A1:D8/log (id: sub-3)
[2025-09-01 20:27:02.369] [debug] [websocket_server.cpp:218] [270] STOMP subscribing to topic nspanel/C0:49:EF:D3:A1:D8/log_backlog (id: sub-4)
[2025-09-01 20:27:02.378] [info] [mqtt_manager_config.cpp:215] [33] US TFT3 GUI file checksum: d48eaf57914a9a5fb2c1fc79283b8967
[2025-09-01 20:27:02.398] [info] [mqtt_manager_config.cpp:225] [33] US TFT4 GUI file checksum: 2b054ff31525b7359d39e6e13dc72b73
[2025-09-01 20:27:02.398] [debug] [websocket_server.cpp:231] [270] STOMP subscription ID sub-4 unsubscribing.
[2025-09-01 20:27:02.423] [info] [mqtt_manager_config.cpp:235] [33] US landscape mirrored TFT1 GUI file checksum: 119ff4e3a1e918069243385770b7dc88
[2025-09-01 20:27:02.448] [info] [mqtt_manager_config.cpp:245] [33] US landscape mirrored TFT2 GUI file checksum: 83780a8b82aa3b69e098131579768f45
[2025-09-01 20:27:02.474] [info] [mqtt_manager_config.cpp:255] [33] US landscape mirrored TFT3 GUI file checksum: 26197411588ecd694eb86962af9404bb
[2025-09-01 20:27:02.499] [info] [mqtt_manager_config.cpp:265] [33] US landscape mirrored TFT4 GUI file checksum: 667087277d851ea1d6b9cd7c7df4eafe
[2025-09-01 20:27:02.502] [info] [entity_manager.cpp:82] [33] Loading config...
[2025-09-01 20:27:02.506] [info] [entity_manager.cpp:165] [33] Loading 1 lights.
[2025-09-01 20:27:02.507] [debug] [light.cpp:54] [33] Loading light 1::Mon Chevet.
[2025-09-01 20:27:02.507] [debug] [entity_manager.cpp:206] [33] Loaded 1 lights
[2025-09-01 20:27:02.507] [info] [entity_manager.cpp:212] [33] Loading 0 buttons.
[2025-09-01 20:27:02.507] [debug] [entity_manager.cpp:253] [33] Loaded 0 lights
[2025-09-01 20:27:02.508] [info] [entity_manager.cpp:259] [33] Loading 0 switches.
[2025-09-01 20:27:02.508] [debug] [entity_manager.cpp:300] [33] Loaded 0 lights
[2025-09-01 20:27:02.508] [info] [entity_manager.cpp:305] [33] Loading 0 scenes.
[2025-09-01 20:27:02.508] [debug] [entity_manager.cpp:344] [33] Loaded 0 scenes
[2025-09-01 20:27:02.509] [info] [entity_manager.cpp:351] [33] Loading 1 global pages.
[2025-09-01 20:27:02.509] [debug] [room_entities_page.cpp:53] [33] Loading RoomEntitiesPage 3, trying to get settings from DB.
[2025-09-01 20:27:02.509] [debug] [room_entities_page.cpp:62] [33] RoomEntitiesPage 3 loaded successfully.
[... cut ...]
[2025-09-01 20:28:38.612] [debug] [nspanel.cpp:389] [51] NSPanel 2::NSPanelChambre loaded 1 rooms.
[2025-09-01 20:28:38.612] [debug] [nspanel.cpp:402] [51] Sending updated NSPanelConfig to panel 2::NSPanelChambre over MQTT.
[2025-09-01 20:28:38.614] [debug] [websocket_server.cpp:285] [51] Updating stomp topic JSON 'nspanel/C0:49:EF:D3:A1:D8/status'
[2025-09-01 20:28:38.614] [debug] [nspanel.cpp:510] [51] NSPanel 2::NSPanelChambre became ONLINE.
[2025-09-01 20:28:38.616] [debug] [websocket_server.cpp:285] [51] Updating stomp topic JSON 'nspanel/C0:49:EF:D3:A1:D8/status'

tpanajott, 2025-09-01:
I'm running up my manager at home right now (dev version) to check a few things but I can't see that the panel sends a request to /download_tft/<panel id> as it should. There should've been a GET-request to that endpoint. Are you sure you have the right IP and port configured in manager address and port?

tpanajott, 2025-09-01:
Also, the panel says it's trying to flash a file that it think is 32673 bytes when in reality the TFT files are more like 15753916 bytes. So, there's probably an issue with the manager address and port settings.

You can test if it's correctly installed.

1. Take note of your manager address and port. (In the example below 192.168.1.5 and port 8080).
2. Enter http://192.168.1.5:8080/checksum_tft_file/2 (2 in this case is the ID of your NSPanel).
3. It should spit out something like "654791ed04b008a31ee43f74b5a70f57". No error or anything, just a simple MD5 checksum.

If there's an error then either the IP or the port is missconfigured.

erich74, 2025-09-01:
ok let me check this...

First I'm absolutely sure the manager IP and port are correct 
HA is running on 192.168.1.126 and the addon is configured on port 8000 (default)
I've got the same in NSPanel manager and I am sure this is the right info because when I trigger a GUI flash it does something on the NSPanel device display (the wrong version info etc...) so it can reach the NSPanel device (but probably with the wrong file as you mention).

I entered the url : http://192.168.1.126:8000/checksum_tft_file/1
and got this: 8209ac6badc6560cb86c38ea795350aa

I only have a single NSPanel so I assume the id is 1

erich74, 2025-09-01:
oh ok I see, during my testing I deleted the Panel and registered it again, that's probably why the ID is 2

erich74, 2025-09-01:
Sorry in the meantime I tried to update the firmware (not the gui) just in case but now my nspanel stays offline.
I need to flash it again with the ttl adapter.
This time maybe I can try using esphome online flasher and the bin of the beta v2. Do you know where I can get the bin file ?

tpanajott, 2025-09-01:
Here you go https://github.com/NSPManager/NSPanelManager/blob/beta/docker%2Fweb%2Fnspanelmanager%2Fmerged_flash.bin

erich74, 2025-09-01:
I just reflashed the NSPanel this time using esphome online flasher and the file you gave me the link.

I did the exact same configuration, then I flashed the NSPanel firmware, then the GUI and now...

[image]

Don't know if it will work but it looks much better....just waiting for the GUI update to complete and I'll report here

tpanajott, 2025-09-01:
Hmm, that's super strange but it looks good!

erich74, 2025-09-01:
yeah you certainly know way better than me :) On another ESP32 project I am involved I had issues using the online flasher vs the PC ESP flasher. Sometimes the online flasher was giving inconsistent result. Not that the device was not working but some feature (like wifi AP) were not working. I have no idea but this was maybe a problem with the NSPanel online flashing tool ...

erich74, 2025-09-01:
...58% holding my breath 😅

tpanajott, 2025-09-01:
Hey! Nice work! Super strange issue. First of all ours users (which I've heard from) where the firmware was corrupt but somewhat working.

Hope you like the project. Lots of things planed!

tpanajott, 2025-09-01:
Well then, thanks again for your patience. The issue seem to be resolved. Would you mind closing it? If there's any questions I'm sure someone on discord can answer. Or simply open another issue.

Good news right before bedtime!

erich74, 2025-09-01:
Yeah I'm gonna sleep as a happy man 😊
Closing now, thanks!

closed 2025-09-01 by erich74