NSPManager/NSPanelManager #326: Rearranging entities is broken

Kaylachuuu, 2025-09-19:
Using drag-and-drop to rearrange entities is broken, it does the following:

[image]

tpanajott, 2025-09-19:
What browser are you using?

Kaylachuuu, 2025-09-20:
I'm using Edge, I just reformatted this laptop and haven't had a chance to install anything else on it yet. It also doesn't work on my Macbook though, using Safari.

aaronouthier, 2025-09-20:
Going to chime in here and ask, just for the sake of ruling out possibilities: Did you add any ad-blockers or do you use anti-virus software that loads a browser extension?

Also, what version of Manager are you using, and what Panel version. Panel version is probably less relevant here, but I ask for completeness.

Kaylachuuu, 2025-09-26:
No ad-blockers and no anti-virus other than whatever comes with Windows.

NSPanel Manager is version: 2.1.2-beta3, panel firmware is version 0.3.666.

closed 2025-09-28 by tpanajott